Pschiee/Perfect-Colour #13: DC motor to become a linear actuator ("pusher")

SilviyaGIvanova, 2019-04-12:
How should the DC motor be mounted so that it pushes against the syringe to dispense paint? Initial design does not seem to account for this because the motor itself would spin instead of going up and down.

SilviyaGIvanova, 2019-04-14:
This is done through a threaded plate which is held onto the top plate of the structure via screw rods. As the motor spins the threaded rod it acts as a linear actuator and moves up and down (pushing against the syringe to dispense paint). The motor itself is constrained from spinning as it is placed between two aluminium plates which only allow it to slide.

LeireFroufe, 2019-04-14:
It works efficiently

SilviyaGIvanova, 2019-04-14:
[image]

closed 2019-04-14 by SilviyaGIvanova